Manuel83/craftbeerpi3 #175: self.get_temp() has accuracy of 1 Degree only

jeichelb, 2018-02-22:
Hello together,

during adaptation of the PID logic I found out, that the function to receive the current kettle temperature self.get_temp() returns an integer value and not a float. 

That means the highly sophisticated PID logic (and any other logic as well) works only with an input accuracy of 1 Degree. This is for me much too less for beer brewing and reliably controlling my kettle.

Does anyone know how to change it to float to have maximum accuracy. 

Thanks a lot!

swimIan, 2018-02-22:
To me get_temp() looks like it returns a float by default. The int() part I see is getting the kettle sensor id.

jeichelb, 2018-02-22:
After the PIDArduino module produced unexpected results, I dumped the inputValue in the calc function of the PIDArduino module into a log file
`self._logger.info('Temp: {0}'.format(inputValue))`

When the GUI shows me 34.875°C, the output shows me only 34. I even tried to add 0.0001 before inputValue to avoid a formatting issue, but inputValue is really 34 as integer. I guess the issue is somewhere deeper than the get_temp() function. I tried to figure that out, but it seems to be too deep in the code to identify for me.