Manuel83/craftbeerpi3 #180: Hysteresis is BROKEN

clearwaterbrewer, 2018-04-05:
remove lines 37 and 38 that have the 'else' statement...  

Let me walk you through it:

-snip-
def run(self):
        '''
        Each controller is exectuted in its own thread. The run method is the entry point
        :return: 
        '''
        while self.is_running():

            if self.get_temp() < self.get_target_temp() - float(self.on):
                self.heater_on(100)
            elif self.get_temp() >= self.get_target_temp() - float(self.off):
                self.heater_off()
            else:
                self.heater_off()
            self.sleep(1)

say setpoint is 82, off=4, on=0
at 80 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 80 < 82-0 --> true turn on
self.get_temp() >= self.get_target_temp() - float(self.off) --> skipped
self.heater_off() --> skipped

at 82 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 82 < 82-0 --> false
self.get_temp() >= self.get_target_temp() - float(self.off) --> 82 >= 82-4 --> true - shut off
self.heater_off() --> skipped

say setpoint is 82, off=-2, on=0
at 80 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 80 < 82-0 --> true turn on
self.get_temp() >= self.get_target_temp() - float(self.off) - skipped
self.heater_off() --> skipped

at 82 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 82 < 82-0 --> false
self.get_temp() >= self.get_target_temp() - float(self.off) --> 82 >= 82-(-2) --> false
self.heater_off()   <<<<----- THIS is turning it off when it should not be!!!!

How I fixed manually:
sudo /etc/init.d/craftbeerpiboot stop
sudo sed -i.bak -e '37,38d' craftbeerpi3/modules/base_plugins/hysteresis/__init__.py
sudo shutdown -r now

lalo-uy, 2018-04-05:
The first example is wrong settings, since you say wanted heater on at 82
and off a 78.
For me is a little confusing the fact that the offsets are subtracted, but
is just a convention.

clearwaterbrewer, 2018-04-05:
NO....
I understand, I found it broken, I have fixed it.

The numbers add just fine, and I know how negatives work, but the "else:" statement on line 37-38  turns off the output when it should be left on.

-mike

clearwaterbrewer, 2018-04-05:
say setpoint is 82, off=-2, on=0
at 80 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 80 < 82-0 --> TRUE -  turn on 
self.get_temp() >= self.get_target_temp() - float(self.off) - SKIPPED
self.heater_off() -->SKIPPED

at 82 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 82 < 82-0 --> FALSE - continue
self.get_temp() >= self.get_target_temp() - float(self.off) --> 82 >= 82-(-2) --> FALSE - continue
self.heater_off()     <<<<----- THIS is turning it off when it should not be!!!!

jfl88, 2018-05-15:
I think for a heater it's assumed you will always have the ON temp less than the OFF temp...  Why would you turn on heater when it's above set point and turn off when it's below setpoint?  That is backwards.

clearwaterbrewer, 2018-05-15:
That is great and all, but without a 'deadband' in the middle, it is not hysteresis...   The last line that has "selfheater_off()" removes the deadband, plain and simple..

lalo-uy, 2018-05-15:
the offsets are note treated equal for heat and for cool.

            if temp + float(self.heater_offset_min) <= target_temp:
>                 self.heater_on(100)
>             if temp + float(self.heater_offset_max) >= target_temp:
>                 self.heater_off()
>             if temp >= target_temp + float(self.cooler_offset_min):
>                 self.cooler_on(100)
>             if temp <= target_temp + float(self.cooler_offset_max):
>                 self.cooler_off()

 As you can see in the heater the offset is added to the actual temp, in
the cooler it is added to the target.

clearwaterbrewer, 2018-05-15:
well, how do you explain that hysteresis does not work, with either offset set to 10 and the other set to zero?

jfl88, 2018-05-15:
> say setpoint is 82, **off=-2**, **on=0**

Why is your OFF less than ON?  You are trying to heat this liquid, correct?  So your OFF should be higher than your ON.  You want your heater to turn ON when the temperature is LOWER than the setpoint and turn OFF when the temperature is HIGHER than the setpoint.  That is how deadband control works.

If other people were having issues with this hysteresis heating then there would be a lot more comments.

clearwaterbrewer, 2018-05-16:
Please show, with your values, how the logic is walked through like I did...

Question, why does the interface not have a more intuitive description of filter the input? (yes, even the mouseover hints are incorrect(both say they should be less than ... )  I can reverse the values and it still does not give actual hysteresis like a Ranco or A419 controller does..

jfl88, 2018-05-16:
OK, let's swap the ON and OFF thresholds then.

**say setpoint is 82, off=0, on=-2**
at 80 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 80 < 82- (-2) --> TRUE - turn on
self.get_temp() >= self.get_target_temp() - float(self.off) - SKIPPED
self.heater_off() -->SKIPPED

at 82 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 82 < 82- (-2) --> TRUE - stay on
self.get_temp() >= self.get_target_temp() - float(self.off) - SKIPPED ( 82 >= 82-(-2) )
self.heater_off() -->SKIPPED

at 84 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 84 < 82- -2 --> FALSE
self.get_temp() >= self.get_target_temp() - float(self.off) --> 82 >= 82-(-2) --> TRUE - turn off
self.heater_off() 

The 'else' statement only gets hit if you have your ON / OFF values set incorrectly.

jfl88, 2018-05-16:
OK, let's swap the ON and OFF thresholds then.

say setpoint is 82, off=0, on=-2
at 80 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 80 < 82- (-2) --> TRUE - turn on
self.get_temp() >= self.get_target_temp() - float(self.off) - SKIPPED
self.heater_off() -->SKIPPED

at 82 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 82 < 82- (-2) --> TRUE - stay on
self.get_temp() >= self.get_target_temp() - float(self.off) - SKIPPED ( 82 >= 82-(-2) )
self.heater_off() -->SKIPPED

at 84 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 84 < 82- -2 --> FALSE
self.get_temp() >= self.get_target_temp() - float(self.off) --> 84 >= 82-(-2) --> TRUE - turn off
self.heater_off()

The 'else' statement only gets hit if you have your ON / OFF values set backwards.

clearwaterbrewer, 2018-05-16:
if off is 0, why are you subtracting (-2)  in the 2nd line of the 2nd two portions of the test walk-through, for 82 and 84???

jfl88, 2018-05-16:
right, brain fart, but even if you replace with the self.off (0) you would get the same outcome...

say setpoint is 82, off=0, on=-2
at 80 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 80 < 82- (-2) --> TRUE - turn on
self.get_temp() >= self.get_target_temp() - float(self.off) - SKIPPED
self.heater_off() -->SKIPPED

at 82 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 82 < 82- (-2) --> TRUE - stay on
self.get_temp() >= self.get_target_temp() - float(self.off) - SKIPPED ( 82 >= 82-(-2) - while TRUE, it does not get checked as it's after the previous condition which is TRUE - elif)
self.heater_off() -->SKIPPED

at 84 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 84 < 82- (-2) --> FALSE
self.get_temp() >= self.get_target_temp() - float(self.off) --> 84 >= 82-(0) --> TRUE - turn off
self.heater_off()

jfl88, 2018-05-16:
Right, brain fart, let's look again:

say setpoint is 82, off=0, on=-2
at 80 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 80 < 82- (-2) --> TRUE - turn on
self.get_temp() >= self.get_target_temp() - float(self.off) - SKIPPED
self.heater_off() -->SKIPPED

at 82 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 82 < 82- (-2) --> TRUE - stay on
self.get_temp() >= self.get_target_temp() - float(self.off) - SKIPPED ( 82 >= 82-(0) also TRUE but not executed as it's an ELSE IF and the IF was true)
self.heater_off() -->SKIPPED

at 82.1 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 82.1 < 82 - (-2) --> TRUE - stay on
self.get_temp() >= self.get_target_temp() - float(self.off) --> 84 >= 82.1 -(0) --> TRUE - turn off (not executed - again, because it's an ELSEIF)
self.heater_off()

at 84 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 84 < 82- -2 --> FALSE
self.get_temp() >= self.get_target_temp() - float(self.off) --> 84 >= 82-(0) --> TRUE - turn off
self.heater_off()

So it looks like if you swap your off and on, that it still won't turn off at the right time with that combination of parameters.

If instead there was a "+" instead of a "-" it looks like the hysteresis would work as expected.

clearwaterbrewer, 2018-05-16:
yep, it is broken...  here is the right example...

say setpoint is 82, off=0, on=-2
at 80 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 80 < 82- (-2) --> TRUE - turn on and exit
self.get_temp() >= self.get_target_temp() - float(self.off) --> 80 >= 84-(0) SKIPPED
self.heater_off() -->SKIPPED

at 82 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 82 < 82- (-2) --> TRUE - stay on and exit
self.get_temp() >= self.get_target_temp() - float(self.off) --> SKIPPED, but TRUE(82 >= 82-(0))
self.heater_off() -->SKIPPED

at 84 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 84 < 82- (-2) --> FALSE, Continue
self.get_temp() >= self.get_target_temp() - float(self.off) --> 84 >= 82-(0) --> TRUE - turn off
self.heater_off()

So, it turns on at setpoint plus 2 (FYI - not intuitive to a cbPi user, not one bit!)

now, when the temp drops back down, where does it turn back on?  
at 83.9 degrees:
self.get_temp() < self.get_target_temp() - float(self.on) --> 83.9 < 82- (-2) TRUE - turn on and exit
self.get_temp() >= self.get_target_temp() - float(self.off) --> SKIPPED
self.heater_off() -->SKIPPED

if it turns on at 84 and back on at 83.9, there is NOT hysteresis...  RIGHT?????

clearwaterbrewer, 2018-05-16:
again, Ranco and A419 are the models to follow if you want a cbPi user to be able to operate it, they are rock simple...

jfl88, 2018-05-16:
The logic just needs to be changed to:
`First condition:
self.get_target_temp() **+** float(self.on)
Second condition:
self.get_target_temp() **+** float(self.off)`

clearwaterbrewer, 2018-05-17:
why does there need to be two settings anyway?  answer, there doesn't...

clearwaterbrewer, 2018-05-17:
say setpoint is 82, off=0, on=-2
at 79 degrees:
self.get_temp() < self.get_target_temp() + float(self.on) --> 79 < 82+(-2) --> TRUE - turn on and start loop again
self.get_temp() >= self.get_target_temp() + float(self.off) --> 79 >= 82-(0) SKIPPED
self.heater_off() -->SKIPPED

at 82 degrees:
self.get_temp() < self.get_target_temp() + float(self.on) --> 82 < 82+(-2) --> FALSE, Continue(stay on)
self.get_temp() >= self.get_target_temp() + float(self.off) --> 82 >= 82-(0) --> TRUE - turn off and start loop again
self.heater_off() -->SKIPPED

at 84 degrees:
self.get_temp() < self.get_target_temp() + float(self.on) --> 84 < 82+ (-2) --> FALSE, 
self.get_temp() >= self.get_target_temp() + float(self.off) --> 84 >= 82+(0) --> TRUE, stay on and start loop again)
self.heater_off()

So
it seems to turn on at any temp 79.999 and below
if on, it turns back off at 82.000
why is the thrid line there????

now, when the temp drops back down, where does it turn back on?  
at 80.0 degrees:
self.get_temp() < self.get_target_temp() + float(self.on) --> 80.0 < 82+(-2) FALSE, Continue(stay off)
self.get_temp() >= self.get_target_temp() + float(self.off) --> TRUE - stay off and start loop again
self.heater_off() -->SKIPPED

at 79.999 degrees:
self.get_temp() < self.get_target_temp() + float(self.on) --> 79.999 < 82+(-2) TRUE - turn on and start loop again
self.get_temp() >= self.get_target_temp() + float(self.off) --> SKIPPED
self.heater_off() -->SKIPPED

so this is hysteresis, but now 'on' has to be less than off...

easier would be to make it like a Ranco or Johnson A419 controller. a differential, and an offset...

lalo-uy, 2018-05-17:
Quote "so this is hysteresis, but now 'on' has to be less than off..."

that make sense, you are heating, you turn on lower than turn off.

clearwaterbrewer, 2018-05-17:
but to get to that point, it is still a change to the code, so just admit that it is in fact broken!

jfl88, 2018-05-17:
@clearwaterbrewer you would get better response if you were a bit more polite about your contributions.  This is a collaborative space, everyone here is contributing their time for free, the software you are using is free.  No one is here to get brow-beaten or yelled at.

Relax, have a beer.

clearwaterbrewer, 2018-05-17:
I politely ask that you spend as much time understanding that it may actually be broken as you do saying I am wrong....  I have quite a few hours in trying to make it better on this post alone...

I want to make it work, and i want to make it easy for cbpi3 users to implement. most have used or have a friend with a Ranco or a A419...

can you admit that is is broken?  can you explain that it is broken to Lalo-uy??? No one is going to fix it if  people are still saying it is not broken....

jfl88, 2018-05-17:
Even the title of this issue is pretty rude - "is BROKEN" is not a great way to start a conversation.

There's no argument here, the logic needs to be changed.  In my opinion the issue is simply a reverse of sign on the comparison but your method would work as well.  What is easier to understand for the layperson homebrewer?  That's debatable, I personally think "ON" and "OFF" is easier to understand than "Offset" and "Differential", but even then the existing tooltips in CBPi don't appear clear at all.

The "Else" case should be kept though, the safest option is to turn a heater off if it's not being used correctly.

clearwaterbrewer, 2018-05-18:
The title is true, and it is a basic feature...

The  else statement breaks it...  see the 3rd of 4 situations in the last scenario I posted, if you have the else in there, it would shut off, which is why it is BROKEN now, and removing it as I show in my post FIXES it...

If you spent more time on the code and less on trying to prove I am wrong, you might have seen that.

You are welcome to spend time writing code, but if you continue posting incorrect statements, you really aren't helping the community.

PR #181 (2018-05-23): Fix hysteresis turning off heater
removed lines 37 and 38 to make hysteresis work properly.  closes #180
closed 2018-05-23 by Manuel83