Manuel83/craftbeerpi3 #183: Feature request: Automatically start wifi hotspot

oleost, 2018-04-08:
Having Craftbeerpi as headless without any screen and relying on WiFi connection can be an hassle. Maybe you gotten new router, new password or in a different location.

Changing wifi password is not easy to do either.

I would suggest an option that if no network is detected it will create its own hotspot.

Philusha1983, 2018-04-08:
Great idea - can be very useful

Philip   ;)

oleost, 2018-04-08:
http://www.raspberryconnect.com/network/item/330-raspberry-pi-auto-wifi-hotspot-switch-internet

oleost, 2018-04-09:
So I followed the install instruction on the link provided.

When I start the RPi3 without ethernet plugged in it succesfully creates an hotspot I`m able to connect to.

But when trying to access Craftbeerpi @ 192.168.50.5:5000 ( Which is the RPi IP address" it fails to connect.

Any idea if this is Craftbeerpi fault or the provided script?

lalo-uy, 2018-04-09:
Check with netstat -lntp where the python process is listening

oleost, 2018-04-09:
[image]

oleost, 2018-04-09:
@lalo-uy This is what netstat gives me.

oleost, 2018-04-09:
So I started and stopped Craftbeerpi and then it works.

But after a reboot now both SSH and Craftbeerpi is refusing connection.

jfl88, 2018-05-16:
This is more of an issue for the operating system.  If you change your Wifi router you can simply turn off the Raspberry Pi, connect the SD card to a computer that can read the filesystem and manually change the settings in the wpa_supplicant configuration.