Manuel83/craftbeerpi3 #184: can't start cbp3

minimalistlux, 2018-04-13:
tried to login in with chromium browser but it won't let me connect
any help is greatly appreciated

HooTsBrooklyn, 2018-04-13:
Try following the directions here.  https://cbpi.mricesolutions.com/viewtopic.php?f=3&p=18

minimalistlux, 2018-04-13:
> Try following the directions here. https://cbpi.mricesolutions.
> com/viewtopic.php?f=3&p=18

Danke

minimalistlux, 2018-04-13:
ty hoots
I figure it out 
I placed cbp3 in the cbp2 directory and that's why it wouldn't start

minimalistlux, 2018-04-14:
but does craftbeerpi always need to be connected to a network? or can I just brew without connecting to the ip

jonrhall, 2018-04-14:
It never needs to be connected to a network, you can run the UI off of the local machine with a screen connected to the raspberry pi, and in that case you'd never be connected to the Raspberry Pi external IP, you'd actually just be going to localhost:5000 on the device's browser itself.

If your question is if CBPi will run when you're not connected to it via the network, yes it will run. It does whatever you've programmed it to do as long as you've configured it properly--but to configure it properly you'll need access to its interface in some way.

minimalistlux, 2018-04-14:
I do connect through localhost:5000 but when I turn off my wifi or disconnect the Ethernet cable the ui doesn't work anymore. 
where can I find help on how to configure to use the cbpi attached to a computer screen without a network.
Is that kiosk mode?
thank you by the way jonrhall

HooTsBrooklyn, 2018-04-14:
Try setting a static ip address for the pi.  Kiosk mode basically starts the browser full screen and disables the right click function.

HooTsBrooklyn, 2018-04-14:
The service runs on the Pi and does not need network/internet access.  You can try using localhost:5000 or 127.0.0.1:5000 on the pi directly.  If you are not on the pi directly, then you would need network access.

localhost and 127.0.0.1 are both loopback addresses

Loopback address is a special IP number (127.0.0.1) that is designated for the software loopback interface of a machine. The loopback interface has no hardware associated with it, and it is not physically connected to a network.

minimalistlux, 2018-04-14:
ok got it to work 
thank you so much for your help hootsbrooklyn