Manuel83/craftbeerpi3 #186: Is craftbeerpi still under active development

sirjhcjw, 2018-04-26:
Hello All,

I am currently building a system and I have identified CBPi3 as the best software for me to use. I am just wondering if it is still under active development and will 3.1 be completed?  I only ask because there have not been many commits lately.

Regards

Chris

LucLN, 2018-05-15:
.. and no official answer to your question ..

HotPepperDaddy, 2018-05-15:
It would help if it was actually open source.

D4p0up, 2018-05-29:
It is apparently, UI is being revamped at the moment.

linuxguy123, 2019-02-13:
Why aren't we writing a CBPi replacement ?   Seems to be enough developers interested in doing something with CBPi and yet we aren't allowed to.  

It is ridiculous that the entire project is controlled by one person.  I'm pretty sure he is doing this because one day he wants to withdraw it from the public domain and take it private.