Manuel83/craftbeerpi3 #187: MAX31865 Dual Channel Code

KenCope, 2018-05-06:
Would anyone be willing to write the code to utilize the Playing With Fusion dual channel MAX31865?  I've tried using their examples without success.  I purchased two of them and cannot get to work :-(

lalo-uy, 2018-05-07:
For the terminal in the picture
http://www.playingwithfusion.com/imgviewer.php?pdid=30&imgid=1157

you have to configure as 2 separated boards, one with CS on the GPIO
connected to CS1 and other with CS2

KenCope, 2018-05-08:
I have it set up that way. I thought it would work just like any other 31865. From what I’m gathering from Playin With Fusion is that it takes a different program.

Thanks.

http://www.playingwithfusion.com/imgviewer.php?pdid=30&imgid=1157

lalo-uy, 2018-05-08:
Tray to set CS2 to 3.3v and connect the other as a normal Max31865 board to
verify.

KenCope, 2018-05-10:
What will applying 3.3V to CS2 do?