Manuel83/craftbeerpi3 #189: iOS web access problem

LucLN, 2018-05-13:
Hi,
when I access my working CBP3 via browser from an iOS device I'm not able to start/stop, modify or open the steps details in the steps-widget. All other widget and browser-access from a computer work.
Does anyone have the same problem or a solution for this?
Luc.

lalo-uy, 2018-05-14:
Is the same on Android. The walk around is to set "show as a Desktop" in
the browser preferences

LucLN, 2018-05-14:
.. so this seems to be a problem in the frame-work ..
Thanks for answering!

perlporter, 2018-05-16:
Does anyone know how we can solve this issue?

DrRataplan, 2018-05-16:
This issue is with the UI code, which is unfortunately not checked in. See #75 for more info.

For now, I use the 'request desktop page' functionality of Chrome for Android when I need to use a popover, and revert back to the mobile view when I just want to look at the status of the brewing process.

HooTsBrooklyn, 2018-05-16:
I just tried to use the fermentation page via mobile and it worked for me.  Dunno if this is a brewing page only issue.

LucLN, 2018-05-17:
yes, for me it's on the brew-page only .. 

(found some other issues in fermentation, with reference to [#186](https://github.com/Manuel83/craftbeerpi3/issues/186) )

closed 2023-02-11 by LucLN