Manuel83/craftbeerpi3 #191: Plugin HTTP-Sensor

mreucker, 2018-05-20:
I use this Plugin. It works fine.
How can i change the unit of the sensor.
It is always celcius degree. I must have unit Liter. Can anyone help?

jonrhall, 2018-05-20:
Hi @mreucker, I don't think it's possible to convert the units to Liters. I believe the sensor logic in craftbeerpi is set to use temperature units like Fahrenheit and Celsius only. These settings are located in the "Parameter" section of the CraftBeerPi3 UI, specified by the "Unit" setting.

Fuzzbinder, 2018-09-26:
Hi @mreucker 
I would like to use this plugin to send temperatures over my  home network to craftbeerpi3.  Can you give me an example of code you use please as I don't have a clue.  Would be using ds18B20s and esp32s or esp8266. Thanks.

mreucker, 2018-09-27:
Hi,
There is the sketch. I’m using a WemosD1.
Regards Matthias

Fuzzbinder, 2018-09-27:
Hi again @mreucker and thanks for responding.  Please excuse me i misunderstand.  Do you mean there is a sketch somewhere?   Cheers Johnnie

mreucker, 2018-09-27:
Hi, have a look.

#include <ESP8266WiFi.h>
#include <DallasTemperature.h>
#include <OneWire.h>
#include "SH1106Brzo.h"

#define ONE_WIRE_BUS D3

SH1106Brzo  display(0x3c, D1, D2);

OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature DS18B20(&oneWire);

const char* ssid = "xxx"; //Hier SSID eures WLAN Netzes eintragen
const char* password = "xxx"; //Hier euer Passwort des WLAN Netzes eintragen

const char* host = "192.168.0.65"; //Server der die temperatur empfangen soll 
const int httpPort = 5000;

const char* script1 = "/api/httpsensor/Temp_1/"; //URL/Verzeichnis das wir gewaehlt haben
const char* script2 = "/api/httpsensor/Temp_2/"; //URL/Verzeichnis das wir gewaehlt haben
const char* script3 = "/api/httpsensor/Temp_3/"; //URL/Verzeichnis das wir gewaehlt haben

int period = 1000;
unsigned long time_now = 0;
 
int CHILLER1_PIN = D6;
int HEATER1_PIN = D7;

int CHILLER2_PIN = D0;
int HEATER2_PIN = D5;

int CHILLER3_PIN = 3;

String TEMP1;
String TEMP2;
String TEMP3;

WiFiServer server(5001);
 
void setup() {
  
  DS18B20.begin(); 

  pinMode(1, FUNCTION_3);
  pinMode(3, FUNCTION_3);
 
  pinMode(CHILLER1_PIN, OUTPUT);
  digitalWrite(CHILLER1_PIN, HIGH);
  
  pinMode(HEATER1_PIN, OUTPUT);
  digitalWrite(HEATER1_PIN, HIGH);

  pinMode(CHILLER2_PIN, OUTPUT);
  digitalWrite(CHILLER2_PIN, HIGH);
  
  pinMode(HEATER2_PIN, OUTPUT);
  digitalWrite(HEATER2_PIN, HIGH);

  pinMode(CHILLER3_PIN, OUTPUT);
  digitalWrite(CHILLER3_PIN, HIGH);
 
  WiFi.mode(WIFI_STA);
  WiFiStart();

  server.begin();

  display.init();

[... 235 lines cut ...]
void Anzeige3 () {
  
  display.clear();

  display.setTextAlignment(TEXT_ALIGN_LEFT);
  display.setFont(ArialMT_Plain_16);
  display.drawString(40, 8, "Chiller");
  display.setFont(ArialMT_Plain_24);
  display.drawString(25, 32, TEMP3 + " °C");
  display.display();
  
}

void WiFiStart()
{
 
  // Connect to WiFi network
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    }

}

Fuzzbinder, 2018-09-27:
@mreucker thankyou so much 👍