Manuel83/craftbeerpi3 #199: Cannot set up OneWire Sensor - address dropdown broken

bitbot-xx, 2018-06-07:
When setting up a OneWire Sensor, the "Sensor" field (with help message "The OneWire sensor address") is a dropdown with a single option "PLEASE SELECT".

[image]

I'm using latest master as of the creation date of this issue.

lalo-uy, 2018-06-07:
Check that the sensors are visible by the OS with:

ls /sys/bus/w1/devices/

bitbot-xx, 2018-06-07:
Thanks @lalo-uy - I don't think my sensor is working as expected... I'm a N00b. It turns out it's an Arduino one with analogue output, not digital... :-/

[image]

However, the UI could be improved by saying "No sensors detected", verifying that the software was indeed installed with 1W support, and linking to some troubleshooting docs?

I'd prefer to leave this issue open until I get a proper 1W sensor and confirm the problem doesn't exist on master. That ok?