Manuel83/craftbeerpi3 #203: Feature Request: Allow HEAD requests on /api/log routes

jonrhall, 2018-07-13:
CraftBeerPi3 logs can grow to very large sizes (read: hundreds of megabytes) very easily, which makes them hard to manage. When you're using the `/api/logs/` endpoint, all you get is a list of log names, and when you want to know which logs have ballooned in size the only way to find out is by downloading them all via HTTP GET--a costly operation. A more efficient way of knowing the log size would be to allow HTTP HEAD requests on the log files themselves.

A sample query might be: 
`curl -I http://<<craftbeerpi-address>>:5000/api/logs/app.log`
... which would return a response with the `Content-Length` header present (which is the size of the file itself), as well as some other useful information like last date modified and date created, all without actually having to download a potentially massive file via a normal GET request.

Thanks for all your work @Manuel83 !

jonrhall, 2018-07-14:
Actually, this issue appears to have been a problem with my debug configuration -- CraftBeerPi3 does allow you to make HEAD requests (`curl -I`) on log endpoints, I just was using the wrong endpoint when looking to make a HEAD request in the first place (instead of `/api/logs/app.log` it should have been `/api/logs/download/app.log`).

closed 2018-07-14 by jonrhall

lalo-uy, 2018-07-15:
I see that the endpoint have a method associated. may be you have to test
and generate a new endpoint for HEAD.