Manuel83/craftbeerpi3 #204: Uninstall

hyper1685, 2018-07-24:
i just installed craftbeerpi3 on my system and the program gets stuck on the load screen in the browser.  i assume the program did not install properly.  how do in uninstall the program so that it can be reinstalled?  I also have craftbeerpi2.2 on my system.  could it be causing the issue?  IF so, how do I unistall that program as well?

TIA

germanherbik, 2018-07-27:
you need stop de program and delete the folder.

lalo-uy, 2018-07-29:
Rub the install.sh and first stop the program, next remove it from starting
on boot. Exit install.sh and remove the folder.

Kywalh, 2018-11-03:
Problem solved !

beardedbrewer, 2019-01-25:
> Problem solved !

How did you solve it? I'm trying to delete it from my RPi to do a fresh install, but I can't figure out how.

jabib, 2020-05-03:
@beardedbrewer I just did that same thing. Did you have to delete everything and start from scratch?

jabib, 2020-05-03:
Nevermind. Managed to fix it with some git commands:
`sudo git checkout master
sudo git pull`
Worked fine again after a reboot