Manuel83/craftbeerpi3 #210: BoilStep doesn't stop

Wobbly74, 2018-08-08:
When running a BoilStep it doesn't ever stop if the kettle temperature is kept above the boil temperature of the BoilStep. This happens particularly with PIDBoil where the element is running at 100% over a set temperature, as opposed to hysteresis or regular PID where the kettle temp may drop below the temp set on the BoilStep due to element cutout.

I traced the problem down to unhandled exceptions in the check_hop_timer methods in the brew_steps plugin.  I adjusted the code in the execute method locally from the standard code:

def execute(self):
        '''
        This method is execute in an interval
        :return:
        '''
        # Check if Target Temp is reached
        if self.get_kettle_temp(self.kettle) >= float(self.temp):
            # Check if Timer is Running
            if self.is_timer_finished() is None:
                self.start_timer(int(self.timer) * 60)
            else:
                self.check_hop_timer(1, self.hop_1)
                self.check_hop_timer(2, self.hop_2)
                self.check_hop_timer(3, self.hop_3)
                self.check_hop_timer(4, self.hop_4)
                self.check_hop_timer(5, self.hop_5)
        # Check if timer finished and go to next step
        if self.is_timer_finished() == True:
            self.notify("Boil Step Completed!", "Starting the next step", timeout=None)
            self.next()

To the following:

    def execute(self):
        '''
        This method is execute in an interval
        :return:
        '''
        # Check if Target Temp is reached
        if self.get_kettle_temp(self.kettle) >= float(self.temp):
            # Check if Timer is Running
            if self.is_timer_finished() is None:
                self.start_timer(int(self.timer) * 60)
            else:
                if len(self.hop_1) > 0:
                    self.check_hop_timer(1, self.hop_1)
                if len(self.hop_2) > 0:
                    self.check_hop_timer(2, self.hop_2)
                if len(self.hop_3) > 0:
                    self.check_hop_timer(3, self.hop_3)
                if len(self.hop_4) > 0:
                    self.check_hop_timer(4, self.hop_4)
                if len(self.hop_5) > 0:
                    self.check_hop_timer(5, self.hop_5)

        # Check if timer finished and go to next step
        if self.is_timer_finished() == True:
            self.notify("Boil Step Completed!", "Starting the next step", timeout=None)
            self.next()

If the length of the hop_# properties is not checked then the check_hop_timer method appears to fail and stop evaluation of the timer check at the end of the method, at least until the kettle temp drops below the BoilStep set point temp (which may never happen, meaning the boil continuing indefinitely until halted manually).

If one of the contributors could test the above to their satisfaction or refactor some other way and then merge back in that would be grand. I'm not a github guy and I've never really played with Python so I'm probably far from qualified to do so :)

Wobbly74, 2018-08-08:
Just to clarify the above, these are unhandled exceptions where these is not value for the hop addition times in the BrewStep config (eg they are blank). If you fill them all in then there is nothing to cause errors of course.

Wobbly74, 2018-08-08:
Slightly more graceful, using isdigit() instead.

    def execute(self):
        '''
        This method is execute in an interval
        :return:
        '''
        # Check if Target Temp is reached
        if self.get_kettle_temp(self.kettle) >= float(self.temp):
            # Check if Timer is Running
            if self.is_timer_finished() is None:
                self.start_timer(int(self.timer) * 60)
            else:
                if self.hop_1.isdigit() == True :
                    self.check_hop_timer(1, self.hop_1)
                if self.hop_2.isdigit() == True :
                    self.check_hop_timer(2, self.hop_2)
                if self.hop_3.isdigit() == True :
                    self.check_hop_timer(3, self.hop_3)
                if self.hop_4.isdigit() == True :
                    self.check_hop_timer(4, self.hop_4)
                if self.hop_5.isdigit() == True :
                    self.check_hop_timer(5, self.hop_5)

        # Check if timer finished and go to next step
        if self.is_timer_finished() == True:
            self.notify("Boil Step Completed!", "Starting the next step", timeout=None)
            self.next()

carlosprite, 2020-07-08:
Just want to leave a note to say thanks, this has been an issue for me 👍🏻