Manuel83/craftbeerpi3 #212: Fermentation steps not start (new user other problem)

germanherbik, 2018-08-14:
Hi, I have the following issue:
Target: 18º
Real temp: 18.5º

Step 1 ---- 19º 5 minutes
Step 2 ---- 14º 5 minutes
Step 3 ---- 21º 5 minutes

(performing a test)

I start but never move to the next step or control the temperature by activating any output!

How are the fermentation steps used?

Thanks in advance.

swimIan, 2018-08-14:
In fermentation the vessel temp needs to be equal to or less than step temp for the steps to proceed.

victorsoliveira, 2020-01-23:
[image]

Why the second step don't move to the next ? The time ends for hours! The first step i had to pass manually to the current step, what's wrong @swimIan ?

Thanks!

lalo-uy, 2020-01-24:
Because time starts when target temp is reached.

victorsoliveira, 2020-01-24:
@lalo-uy But, temp need to be exactly equals a target temp ? 22 is different of 22.120, 22.240, etc ? And because of that step don't start timer ?

If was, what's the solution for decimal variant ?

Thanks!

lalo-uy, 2020-01-24:
The target must be reached once. Look

victorsoliveira, 2020-01-24:
Understood @lalo-uy !

Thanks a lo!

victorsoliveira, 2020-02-12:
@lalo-uy I think i found a problem!

The target temp don't consider the offset, so, if i use 0,5 as offset and my target temp is 19 then the system need to consider as "reached" 19,5 to start timer and not just target temp, 19.

I think so today the system not considering (target temp + offset) to calculate rached temp.