Manuel83/craftbeerpi3 #215: Web page to play audio alerts.

Dubzie, 2018-09-02:
Can we have an option to upload a sound file, which the webpage will play when there is an alert.
I'd like the laptop that displays CBPi to do the alerting rather then having to wire in a buzzer.
Or both, so if i'm not within range of the buzzer i can hear it on a laptop or something like that.

Philusha1983, 2018-09-02:
I made a similar proposal several months ago. at a Facebook post in the
global user group
<https://www.facebook.com/groups/craftbeerpi/permalink/1787065188263290/>
Here is my proposal;

*Voice Notifications Plugin request.*
General concept:
To use a text to speech API service to generate the voice for user-defined
fields during the brew day.
link to a URL API example that I found: http://bit.ly/cbpi_voice
This can be similar to toggle step plugin (small tasks between steps)
Or it can be integrated into any brew day step as a custom field.
Since a text to speech services require a constant internet connection, I
think it's better to generate the voice file after brew step is saved and
played it each time the step is triggered without a constant internet
connection.

Voice Notification Example at brew boss system:
https://youtu.be/9DEjuRk8w0E?t=6m10s
Luke Mullan
<https://www.facebook.com/lmullan89?fref=gc&dti=1620423084927502&hc_location=ufi>
already made a sound related plugin for CBPI so maybe it's a good start.
https://github.com/jalim/cbpi-wav-buzzer
And the last reference is a demo that I made for possible steps that can be
cool to have.
http://bit.ly/brewsteps_demo

Philip   ;)

Dubzie, 2018-09-03:
how do i add that wav buzzer! that exactly what i need!

JamFfm, 2018-10-06:
In command line use: 

git clone https://github.com/jalim/cbpi-wav-buzzer.git -b master --single-branch /home/pi/craftbeerpi3/modules/plugins/cbpi-wav-buzzer

As far as I understand this Plugin can be used as a replacement of the buldin Buzzer and is played from the sound-hardware of the RasPi.

adomax111, 2020-01-24:
To add sound to the browser, do the following:
in the file home /pi/craftbeerpi3/modules/ui/static/bundle.js Add the code after the fragment "var e = this.props.messages;":
var audioAlert = new Audio ("static / beep.wav");
audioAlert.play ();
And put any wav file in the folder home/pi/craftbeerpi3/modules/ui/static/ called beep.wav "

JamFfm, 2020-01-24:
This is very cool!! Pay attention to skip the spaces. For me it works like this: 
.....var e=this.props.messages;var audioAlert=new Audio ("static/beep.wav");audioAlert.play();return d.default.createElem....
Thank you very much!!