Manuel83/craftbeerpi3 #226: Add_ons not populating

clandorsdx, 2019-03-08:
In 3.0 when I go to the add_ons tab under system the plugins don't populate.  I saw that there was an issue with this earlier but it was marked as solved.

msfuture, 2019-03-25:
Same her... fresh clone from git, no addons are loaded. The javascript console prints a 500 Internal Error while loading the plugin list?

HooTsBrooklyn, 2019-03-25:
Does your rpi have internet access?  The plugins list is populated by a file on github.

msfuture, 2019-03-25:
Yes of course. git pull worked like a charm, and the version information page is pulling data as well.

bitterroots, 2019-04-07:
Same here. I have reinstalled and cleared data multiple times.

ShintoTuna, 2019-04-08:
i resolved this issue by reinstalling my Raspbian OS to Jessie version. it seems that Stretch has issues.