Manuel83/craftbeerpi3 #231: alternatives for wiringPi

perlporter, 2019-08-06:
git.drogon.net is temporarily unavailable. Please look for alternatives for wiringPi, etc. -Gordon
http://wiringpi.com/wiringpi-deprecated/

Manuel83, 2019-08-06:
Valid point! Any suggestion for an alternative?

perlporter, 2019-08-06:
Unfortunately no, I'm sorry if I expressed myself the wrong way.
They are words of the creator, not mine ...
I'm just reporting because I can't do a new install.

Manuel83, 2019-08-06:
The question is whether wiringpi is really required. Did you try to select "No" for wiringpi during the installation?

perlporter, 2019-08-06:
No I did not try.
Good to know, I will do the installation doing what you recommended.
I thought it was necessary to have wiringPi installed for the proper functioning of craftbeerPi.

closed 2019-08-06 by perlporter

Manuel83, 2019-08-06:
It’s working?

perlporter, 2019-08-07:
The installation was completed successfully. 
But I haven't tested it yet