Manuel83/craftbeerpi3 #244: Buzzer wont work on CBPi V3.0 board

aydo01, 2020-03-13:
I cant get the buzzer to sound when i have added hop additions. I have tried changing the GPIO in the settings from 16, 20 and 25 and none of them work.

Is there a way i could check on the board what it is meant to be or something like that?

Is there some code you can write into terminal to test the buzzers function?

Cheers

cherryphilip74, 2020-03-13:
Try with gpio5 or gpio6..

lalo-uy, 2020-03-13:
You can create an actor GPIOSimple on the test pin and tourn it on & off to see the buzzer harware working.

stiansolbo, 2020-06-02:
I had the same issue. Buzzer is GPIO 22 by trial and error

lazarosatiro, 2020-06-12:
Hi Guys.

I have this problem,  use the terragady interface my buzzer stay on gpio 25, i'dont view this gpio in parameters, somebody was resolve this problem?