Manuel83/craftbeerpi3 #245: ADC help

cat007mine, 2020-03-15:
Hi all,

I need some simpleton explaining as I'm trying to use my BrewTroller's bubbler volume measuring with CBP3. I haven't played in Linux of C++ (BrewTroller runs on this) for a long time and can be considered a n00b haha.

The BT bubbler system uses an analogue pressure sensor and a fish tank air bubbler with a dip tube into the vessel you're measuring. The pressure of the water creates backpressure in the air line that the pressure sensor detects and sends a signal back.
The pressure sensor however is analogue, and with the Rpi3+ that I'm using only have digital GPIO pins, I have bought a ADS1115 ADC.
I'm assuming that I follow this below wiring guide to wire it into the Pi:

https://learn.adafruit.com/adafruit-4-channel-adc-breakouts/signal-connections

But then I'm not sure how or what to edit on CBP3 to not only register the new input and display it on the dashboard, but also how to configure it so XX signal = a particular volume.
Also the BT has a smoothing/averaging configuration as the signal coming from the sensor looks a bit like a wave.

Any help would be greatly appreciated – and if there’s a guide that can be made for others – this would make the CBP3 (and maybe 4) a VERY powerful home brewing tool!

Thanks!
Hunt

lalo-uy, 2020-03-16:
There is a pressure sensor plugin you can look and modify.
I take a different approach to implement NTC thermometer, now under tewsting. 
I write a program, in any language, that read the ADS1115, do the math and output the value. 
Then user a modified version of the SystemTemp plugin to invoke that program.

cat007mine, 2020-03-16:
> 
> 
> There is a pressure sensor plugin you can look and modify.
> I take a different approach to implement NTC thermometer, now under testing.
> I write a program, in any language, that read the ADS1115, do the math and output the value.
> Then user a modified version of the SystemTemp plugin to invoke that program.

I think I found the plugin, is it this one?

https://github.com/netanelbe/cbpi_pressure_sensor

I've installed it - but have no idea what now haha