Manuel83/craftbeerpi3 #250: pip is not installed by install.sh

nufer, 2020-04-14:
install.sh does not install pip, but relies on it. The error messages are just ignored if the installation fails because of this.

the install.sh should include the setup for pip

hgiachetti, 2020-04-15:
Hi nufer try manual installing:
sudo apt install python-pip

I am trying to install CBPi3, but web is not working.

staticdev, 2020-04-19:
Good it solved my problem! Thanks @hgiachetti .. @Manuel83 should also fix requirements.txt. Current version is not working.

soinwi, 2020-06-18:
I can confirm that updating the GitPython Version from 2.1.3 to 2.1.15 fixed the installation on my system too.
But I guess python-pip was preinstalled on my system (I used a newly installed raspbian, the current version from the "Raspberry Pi Imager")