Manuel83/craftbeerpi3 #252: No IP address

gbucey, 2020-04-29:
I have git cloned CraftBeerPi3 and everything seems to have pulled down without error. Have sudo install.sh and run the installation without issue. When I start the server however, it displays without an IP address. The RPi has a valid IP and is functioning normally as I am currently SSH'ed into it.

See screenshot attached. Please advise.

[image]

HooTsBrooklyn, 2020-04-29:
you should be able to access the pi from another computer using the pics IP followed by :5000.

The install script does not pull the pi's IP.

If you are on the pi itself, you should be able to access it via http://localhost:5000

gbucey, 2020-04-29:
Thank you for the recommendation.

We cannot access the CraftBeerPi system from a remote system of the localhost. We are including the port number :5000 after the address. Just shows site cannot be reached.

HooTsBrooklyn, 2020-04-29:
The install may have missed a few pip dependencies... 

This has a nice install guide that helps correct a lot of the troubles. 

https://techdout.dev/2019/09/24/installing-craftbeerpi/

gbucey, 2020-04-29:
Thanks HooTsBrooklyn, I literally just fouind that page through the CBPi subreddit a few minutes ago. Fairly certain this is my problem due to the error message below:

gitdb requires Python '>=3.4' but the running Python is 2.7.16

gbucey, 2020-04-29:
Thanks HooTsBrooklyn. That article did the trick. It was all the missing pip dependencies that was causing CBPi to fail at launch.