Manuel83/craftbeerpi3 #253: Installing craftbeerpi3 print "INIT DB" Error

greidyone, 2020-05-06:
I am trying to install Craftbeerpi3 and I am getting the error. When running ./run.py

Traceback (most recent call last):
  File "./run.py", line 3, in <module>
    from modules import socketio, app, cbpi
  File "/home/pi/craftbeerpi3/modules/__init__.py", line 50
    print "INIT DB"
                  ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("INIT DB")?

adantoft, 2020-05-06:
@greidyone check out this guide: https://techdout.dev/2019/09/24/installing-craftbeerpi/ to make sure it's installing right. requirements.txt is currently broken.

greidyone, 2020-05-06:
@adantoft I went through the guide and can't seem to get past this **print "INIT DB"** Issue.
I have tried changing the line to **print("INIT DB")** but it just goes further and errors out. 
 I am very new to this stuff. I will get a new SD card and start from scratch. 

Thanks for the help

HooTsBrooklyn, 2020-05-06:
His issue is that cbpi is written for python 2.7, and he trying to use python 3.

greidyone, 2020-05-06:
Ok now I have got past the python issue and now I am getting another error. 

I have uninstalled and installed flask_socketio many times and I still get 

Would it need an older version of socketio??

Traceback (most recent call last):
  File "run.py", line 2, in <module>
    from modules import socketio, app, cbpi
  File "/home/pi/craftbeerpi3/modules/__init__.py", line 11, in <module>
    from app_config import *
  File "/home/pi/craftbeerpi3/modules/app_config.py", line 7, in <module>
    from flask_socketio import SocketIO, emit
ImportError: No module named flask_socketio

------------------
(program exited with code: 1)
Press return to continue

HooTsBrooklyn, 2020-05-06:
You're missing pip dependencies, this guide will help. 

https://techdout.dev/2019/09/24/installing-craftbeerpi/

greidyone, 2020-05-07:
Thanks, I have it working now. The two things I had to do is make sure I was running Python 2.7.
This where I changed the default python to be 2.7
https://raspberry-valley.azurewebsites.net/Python-Default-Version/

Then I had to load the proper Dependencies as below. The was more that these two but they where ones that I had to search for a different name than what the module itself is called in the code.
pip install gitpython
pip install PyYAML

Thanks Everyone.

danluckner, 2020-10-11:
Thanks @greidyone!
Your instruction worked for me.