Manuel83/craftbeerpi3 #255: BUG BUZZER I can't set correct gpio

lazarosatiro, 2020-06-15:
Hello,

Use an interface board to control my process (TERRAGADY V5.0), as my beep is soldered at gpio 25. When you can access parameters> the beep, you can select as gpios 16 to 23.

What makes me more intrigued is the update script (which releases folders, but is not working.

I'm using a raspberry pi4.

I have already installed the software and updated several times without success.

Anyone who can help me will be grateful.

lalo-uy, 2020-06-15:
ChecK this command:   sqlite3 craftbeerpi.db "select * from config where
name='buzzer'"
mine say:
buzzer|16|select|Buzzer
GPIO|[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27]

if not you can modify craftbeerpi.db with sqlite3

lalo-uy, 2020-08-10:
You can as root if craftbeerpi is not running.