Manuel83/craftbeerpi3 #261: Community Power

budda85, 2020-09-29:
Guten Morgen Manuel,
ich habe dieses wundervolle Projekt gefunden da ich gerade mit dem Brauen anfangen möchte.
Das Projekt bietet, so wie ich es sehe, sehr viel potenzial. 
Damit es nicht einschläft und es auch mit der Hilfe von der Community weiterentwickelt werden kann,
wäre es schön wenn du,

- die Lizenz anpassen könntest, damit alle ordentlich mitwirken können.
- und du Maintainer mit in das Projekt einbeziehen könntest.

	
Programmieren nimmt ordentlich Zeit in anspruch, vorallem wenn man das in seiner Freizeit macht.
Wäre schön wenn deine ganze investierte Zeit nicht umsonst wäre und lass uns als Community mitmachen.
Beim überfliegen der Issues sind ja schon die ein oder anderen dabei die gerne aktiv mitmachen wollen.

Gruß
Jan

English Version (translated with DeepL ;-)  )

Good morning Manuel,
I have found this wonderful project because I am just about to start brewing.
The project offers, as I see it, a lot of potential. 
So that it doesn't fall asleep and can be developed further with the help of the community,
it would be nice if you,

- adjust the license so that everyone can participate properly.
- and you could include maintainers in the project.

	
Programming takes a lot of time, especially if you do it in your spare time.
It would be nice if all your invested time would not be for free and let us participate as a community.
When browsing through the issues there are already some people who want to actively participate.

Greeting
Jan

mastnym, 2020-10-12:
Many have tried @budda85, no one succeeded. Such a nice project...

Jocke-G, 2020-11-30:
+1
Would be great!

jshuntley, 2021-01-04:
Well said!

LucLN, 2023-11-06:
.. wir sollten alle mit dem Projekt abschliessen .. da kommt nix mehr .. wahrscheinlich ist der Herr Manuel verstorben oder so ..

jabib, 2024-01-29:
I used to have CBPi3 installed, but with all the changes to Python over the years, it seems like more hassle than it's worth to continue to use CBPi3. I just switched to CBPi4 https://github.com/PiBrewing/craftbeerpi4 and not only is it being actively developed by the community in that fork, but it's much easier to do a fresh install with an .iso image file you can flash to card.
https://openbrewing.gitbook.io/craftbeerpi4_support/