Manuel83/craftbeerpi3 #265: Raspberry Pi shuts down when accessing craft beer pi home page via port forwarding

Steven-Gassert, 2021-08-16:
### Description
Perhaps this is a raspberry pi security feature, but whenever I access the crafbeerpi UI on localhost via port forwarding from an outside network, my raspberry pi immediately crashes. Was curious if anyone would be able to provide an explanation for this behavior or provide some setups they have used to access craftbeerpi from outside the network it is connected to.

### My flow
Outside network --> My wifi router port forwarding to raspberry pi --> nginx running on raspberry pi --> craftbeer pi running on localhost

lalo-uy, 2021-08-16:
Why do you put nginx in the middle.?

Steven-Gassert, 2021-08-16:
@lalo-uy My idea was that I would be able to forward requests to several applications running on my local network.

In the past I've also seen this behavior while port forwarding via [ngrok](https://ngrok.com/)

#### Similar flow

Outside network --> ngrok --> craftbeerpi running on localhost

I'll update the description with this additional info

lalo-uy, 2021-08-18:
I prefer to open a vpn to the raspberry.
You can use OpenVPN, and lately ZeroTierOne as a very easy setup.
That puts your remote device in the same network of the raspberry
letting you access any app.
If raspberry Ip forwarding is enabled you even reach your home network.

Lagerman, 2021-08-18:
I use VNC. Easy setup. Works.

Steven-Gassert, 2021-08-19:
Thanks for the replies guys. I do currently use VNC when connecting to my pi. 

My original idea was that I would be able to share a URL with non technically savvy family members who would be able to access the craft beerpi UI remotely (perhaps behind an authentication prompt) hence the preference to avoid some of the existing solutions.

The really strange part is that when I do finally make a connection to the local host via the networking paths above, the entire pi shuts down and I am no longer able to access it via ssh or vnc. That’s why my original thought was that perhaps this was some sort of security feature to prevent outside traffic from gaining control of gpio pins ect